stm32-rs/stm32l0xx-hal #1: Crates.io release

dzarda, 2019-02-26:
First, thank you so much for your work!

I'd like to use this crate, but:
1. Adding it as a `dependency` with `stm32l0xx-hal = { git = "https://github.com/stm32-rs/stm32l0xx-hal.git" }` fails due to `path = "../stm32-rs-arko/stm32l0"`.
2. It's not present on crates.io

#1. should be more straightforward. Can we directly link the crates.io version? What's missing?

arkorobotics, 2019-03-03:
This is all new to me too! If you wish to use this repo, you will need to download and reference [my fork of stm32-rs](https://github.com/arkorobotics/stm32-rs) in your cargo.toml until the next version of stm32-rs (>0.6.0)... though I bet you already figured this out.

Just a heads up, this repo has not been tested on hardware just yet. Everything is a work in-progress, so if you have any tested code, feel free to submit a pull request! Thanks!

ejpcmac, 2019-04-23:
@arkorobotics Are the changes from your fork merged upstream yet—and so just waiting for the upcoming 0.7.0 release?

Also, I can confirm as of today that timers and GPIOs *do* work on a real (custom) STM32L0-based board. I’ve tried a little RTFM blinky app using timers at work.

arkorobotics, 2019-04-23:
Yep, just waiting on the 0.7.0 release so that we can build it without having to checkout my specific fork.

Awesome! Yes, I did test the GPIO, timer, and serial modules on my stm32l0 board. I'm hoping to get 100% hardware tested coverage on this HAL by the end of the month (fingers crossed).

dzarda, 2019-04-28:
I'm having some linker-related issues (you might be too 70bf58a5cab56aff742f240c91f04f3effc4383c)

Tried modifying the linker flags in both HAL and the target project, but to no avail. In some cases, it compiles with zero binary size, which is not idea...

dzarda, 2019-04-28:
```
error: linking with `arm-none-eabi-ld` failed: exit code: 1
  |
  = note: "arm-none-eabi-ld" "-L" "C:\\Users\\Dzarda\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\lol-7759caab3eb22c53.lol.8n7niwze-cgu.0.rcgu.o" "-o" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\lol-7759caab3eb22c53" "--gc-sections" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\debug\\deps" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\build\\lol-23f82ee7a9bcbca8\\out" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-7ad15540ee137b18\\out" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-d76733916a6f6ae6\\out" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-dca21d5558ac41ea\\out" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-semihosting-09218d5b4ee9cc8a\\out" "-L" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\build\\stm32l0-2703d6be609604e1\\out" "-L" "C:\\Users\\Dzarda\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libcortex_m_rt-a8bb61ba359c9ce3.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libr0-0e442d1d6b3879ed.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libcortex_m-43873eefa1219f85.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libcortex_m-baea52a4b9c7c813.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libaligned-bd05f62d8aa78850.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libas_slice-e42f9c57c1087dbc.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libstable_deref_trait-6096349676a01e7e.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libgeneric_array-ac3723ed70ea01ab.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libtypenum-a87e1d98055b0522.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libvolatile_register-bc85882bb11ec697.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libvcell-e0ea354adc6c77fc.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libbare_metal-1e0243ddac44e6e9.rlib" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libaligned-963c190bdfbe9a77.rlib" "--start-group" "C:\\Users\\Dzarda\\Documents\\src\\lol\\target\\thumbv6m-none-eabi\\debug\\deps\\libpanic_halt-89eacd39240a076c.rlib" "C:\\Users\\Dzarda\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\librustc_std_workspace_core-8622d0c81072f188.rlib" "C:\\Users\\Dzarda\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcore-5d7801e9a874ff0a.rlib" "--end-group" "C:\\Users\\Dzarda\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcompiler_builtins-65d8eda8fd6636ea.rlib" "-Tlink.x" "-Bdynamic"
  = note: C:\Program Files (x86)\GNU Tools ARM Embedded\8 2018-q4-major\bin\arm-none-eabi-ld.exe:
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
```

arkorobotics, 2019-04-28:
Can I ask you to post your `cargo.toml` file? Do you have the latest commit? 

This seems similar to: [https://github.com/stm32-rs/stm32-rs/issues/63](https://github.com/stm32-rs/stm32-rs/issues/63)

arkorobotics, 2019-04-28:
> First, thank you so much for your work!
> 
> I'd like to use this crate, but:
> 
> 1. Adding it as a `dependency` with `stm32l0xx-hal = { git = "https://github.com/stm32-rs/stm32l0xx-hal.git" }` fails due to `path = "../stm32-rs-arko/stm32l0"`.
> 2. It's not present on crates.io
> 
> #1. should be more straightforward. Can we directly link the crates.io version? What's missing?
> 
> EDIT: I see, it's all being worked on right now, big time. Let me know how can I help.
> EDIT2: I managed to use the code anyway, please bear with me, this is a kind of a first for me.

dzarda, 2019-04-29:
Thank you, it's alright. Adding `use stm32l0xx_hal;` was enough :D

arkorobotics, 2019-04-29:
Excellent! 
I'm going to close this ticket since both issues mentioned in the first post are resolved:

1) Resolved by commit [4c10afa](https://github.com/stm32-rs/stm32l0xx-hal/commit/4c10afa3a488adf9316184cfc8d81c14e89ebee6) after the release of stm32-rs v0.7.0 

2) stm32l0-hal v0.1.0 is now available on crates.io! [https://crates.io/crates/stm32l0xx-hal](https://crates.io/crates/stm32l0xx-hal)

closed 2019-04-29 by arkorobotics