stm32-rs/stm32l0xx-hal #3: Flashing method

ejpcmac, 2019-04-22:
I see you’ve added to the `README.md` a flashing method that uses `st-flash`. Any reason not to do this instead?

    $ ./openocd_program.sh target/thumbv6m-none-eabi/debug/examples/serial

arkorobotics, 2019-04-23:
Great question. Up until now I hadn't considered it. I modified the openocd config file (https://github.com/stm32-rs/stm32l0xx-hal/blob/master/openocd.cfg) to attempt this, but I keep running into issues either erasing or flashing the device. Any thoughts as to what the issue might be? 

Output when attempting to only flash:
`arkos-MacBook:stm32l0xx-hal arko$ ./openocd_program.sh target/thumbv6m-none-eabi/debug/examples/serial
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.215810
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
adapter speed: 300 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x1ff00288 msp: 0x200007a8
Info : Device: STM32L0xx (Cat.1)
Info : STM32L flash size is 16kb, base address is 0x8000000
flash 'stm32lx' found at 0x08000000
adapter speed: 300 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x1ff00288 msp: 0x200007a8
STM32L0: Enabling HSI16
adapter speed: 2500 kHz
** Programming Started **
auto erase enabled
Error: error writing to flash at address 0x08000000 at offset 0x00000000
** Programming Failed **
shutdown command invoked
`

Output when attempting a mass erase before flashing:
`
arkos-MacBook:stm32l0xx-hal arko$ ./openocd_program.sh target/thumbv6m-none-eabi/debug/examples/serial
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.211067
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
adapter speed: 300 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x1ff00288 msp: 0x200007a8
Info : Device: STM32L0xx (Cat.1)
Info : STM32L flash size is 16kb, base address is 0x8000000
stm32lx mass erase failed`

ejpcmac, 2019-04-23:
I’ve had some weird issues too with the Nucleo-F767ZI. I’ve noticed that if you wait for the drive to appear, doing nothing with the board in the mean time, it then works reliably. I don’t know if it is the same with STM32L0 dev board though. I’ll receive one this week so I’ll be able to try.

arkorobotics, 2019-07-05:
Resolved the issue in commit [14a318a ](https://github.com/stm32-rs/stm32l0xx-hal/commit/14a318a364a9cc0e18169acc50933ab9059295d0). As it turns out, OpenOCD release version 0.10.0 does not fully support the stm32l0 family. There are several errors in the stm32l0.cfg file that prevent openocd from programming/debugging the MCU. Later commits of OpenOCD resolved this issues, so I added instructions on how to install the newer builds. I also updated the flashing and debugging instructions to include GDB and GDBPy+Dashboard.

closed 2019-07-05 by arkorobotics