stm32-rs/stm32l0xx-hal #41: Add DMA features

berkowski, 2019-09-23:
I'd like to propose a few new features for DMA ops and I'd be happy to start working on a PR. Right now I'd like to use DMA to send data from a buffer I format messages in to a UART.  

- I'd like a `write_some(..)` equivalent to the `write_all(..)` methods on `Tx<UART>` so I can specify how many bytes to write without messing around with the Pin object.   

Eventually I'd like to use DMA with SPI and ADC targets, and with the ADC I'd like to use circular writes.  Neither looks like too much work to add, it just needs to be done.

But with more DMA targets and options, would it make sense to move the methods into traits?

arkorobotics, 2019-09-23:
@hannobraun thoughts?

hannobraun, 2019-09-24:
@berkowski 
> I'd like a `write_some(..)` equivalent to the `write_all(..)` methods on `Tx<UART>` so I can specify how many bytes to write without messing around with the Pin object.

Sounds sensible and straight-forward to implement.

> Eventually I'd like to use DMA with SPI and ADC targets, and with the ADC I'd like to use circular writes. Neither looks like too much work to add, it just needs to be done.

Adding DMA support to SPI shouldn't be too hard. The special thing about SPI is, that you can read and write at the same time, which means you'd have two active DMA transfers working on the same buffer. I suggest adding a `Transfer` struct to the SPI module that wraps the two DMA transfers and basically looks like a single transfer to the user. I've done exactly this before, in another HAL: https://github.com/stm32-rs/stm32f7xx-hal/blob/master/src/spi.rs

Regarding ADC, I'm currently under contract to implement exactly what you described (among other things). I expect to start this work within the next two weeks, so unless you're in a particular hurry, you can just wait for me to do it. If you want to start working on this yourself, I suggest we coordinate here.

> But with more DMA targets and options, would it make sense to move the methods into traits?

I don't know, maybe? I can't really think of a use case that would need to abstract over DMA transfers specifically (which doesn't mean there aren't a million of them). It would make sense to have asynchronous equivalent's of Rust's `Read`/`Write` traits (like Tokio has), but defining them might be beyond the scope of this specific crate.

berkowski, 2019-09-24:
@hannobraun 

All seems reasonable.  I'll put together the `write_some()` PR hopefully soon and then look at the SPI stuff you linked.  I don't have any needs blocking on the ADC stuff so there's no rush for me there.  I just assume that at some point I'll want it when an idea comes along.

I don't have a compelling case for moving on traits for DMA and was mostly just curious if there was one but the effort just hadn't been put on yet.

arkorobotics, 2019-09-25:
@berkowski does #43 satisfy this request? If so, I'll close it.

berkowski, 2019-09-26:
I think so!  At least the work is being done to resolve it, so I'd say it can be closed

arkorobotics, 2019-09-26:
Great, thanks!

closed 2019-09-26 by arkorobotics

hannobraun, 2019-10-14:
Update on my ADC work: I've started working on it, but have to switch to other work for now. I expect that I'll be able to pick it back up some time in late November.

Here's my WIP branch: https://github.com/braun-embedded/stm32l0xx-hal/tree/adc

berkowski, 2019-10-21:
That's great!   I've also gotten sucked back into work that pays the bills so my own little push is on hold, but I'll take a look at that link for my own curiosity's sake

hannobraun, 2020-03-25:
Getting back to the ADC work took me much longer than expected, but I've done it now: https://github.com/stm32-rs/stm32l0xx-hal/pull/86