stm32-rs/stm32l0xx-hal #77: I2C Example - Error Reading Variable

DanSGiesbrecht, 2020-02-13:
I'm having difficulty implementing the I2C example (MCU: STM32L071):

```
#![no_std]
#![no_main]

extern crate panic_semihosting;

use cortex_m_rt::entry;
extern crate stm32l0xx_hal;
extern crate shtcx;
use stm32l0xx_hal::{pac, prelude::*, rcc::Config};

#[entry]
fn main() -> ! {
    let periph = pac::Peripherals::take().unwrap();

    let mut rcc = periph.RCC.freeze(Config::hsi16());
    let gpioa = periph.GPIOA.split(&mut rcc);

    let sda = gpioa.pa10.into_open_drain_output();
    let scl = gpioa.pa9.into_open_drain_output();

    let mut i2c = periph.I2C1.i2c(sda, scl, 100.khz(), &mut rcc);

    let mut buffer = [0u8; 2];
    const ADDRESS: u8 = 0xFF;

    i2c.write(ADDRESS, &mut buffer).unwrap();

    loop {
    }
}
```

---

When debugging using GDB, I get one out of a variety of errors when I stop debugging, such as:

```
^C
Program received signal SIGINT, Interrupt.
0x080009d8 in stm32l0::stm32l0x1::i2c1::isr::<impl stm32l0::generic::R<u32,stm32l0::generic::Reg<u32,stm32l0::stm32l0x1::i2c1::_ISR>>>::txe (
    self=<error reading variable: Cannot access memory at address 0x20004e3c>)
    at /home/gies/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32l0-0.9.0/src/stm32l0x1/i2c1/isr.rs:801
801	        TXE_R::new((self.bits & 0x01) != 0)
```
```
^C
Program received signal SIGINT, Interrupt.
0x080009e8 in stm32l0xx_hal::i2c::I2c<I,SDA,SCL>::send_byte (
    self=<error reading variable: Cannot access memory at address 0x20004dd4>, 
    byte=<error reading variable: Cannot access memory at address 0x20004dd8>)
    at /home/gies/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32l0xx-hal-0.5.0/src/i2c.rs:183
183	        while self.i2c.isr.read().txe().bit_is_clear() {}
```
```
^C
Program received signal SIGINT, Interrupt.
core::cell::UnsafeCell<T>::get (
    self=<error reading variable: Cannot access memory at address 0x20004db0>)
    at /rustc/4560ea788cb760f0a34127156c78e2552949f734/src/libcore/cell.rs:1539
1539	/rustc/4560ea788cb760f0a34127156c78e2552949f734/src/libcore/cell.rs: No such file or directory.
```
```
^C
Program received signal SIGINT, Interrupt.
0x080009bc in vcell::VolatileCell<T>::get (
    self=<error reading variable: Cannot access memory at address 0x20004e38>)
    at /home/gies/.cargo/registry/src/github.com-1ecc6299db9ec823/vcell-0.1.2/src/lib.rs:32
32	        unsafe { ptr::read_volatile(self.value.get()) }
```

---

I'm relatively new to Rust. Is this an issue with the I2C example, or am I missing something else? Any assistance would be greatly appreciated.

dbrgn, 2020-02-13:
@DanSGiesbrecht see #39. we're also using the STM32L071 and hit the same strange issues :slightly_smiling_face:

DanSGiesbrecht, 2020-02-13:
Thanks for pointing that out! Going to test tonight - if that's my only issue, I'll close this PR.

closed 2020-02-15 by DanSGiesbrecht