stm32-rs/stm32l0xx-hal #79: PWM interface only exposed for TIM2

berkowski, 2020-02-15:
I'm looking at using this crate for a device built around the STM32L071 and found that the PWM interface is only exposed for TIM2.  I'll need PWM on TIM3 as well and looked into making a PR when I noticed things were pretty hard-coded in `pwm.rs`. 

Is there any reason things are the way they are with the current implementation? What kind of changes would be allowed to provide a more generic creation of PWM interfaces for the other timers?  

I also have an idea of adding category features to enable specific peripherals following the tables in the reference manuals such as Table 2 in [RM0377](https://www.st.com/content/ccc/resource/technical/document/reference_manual/21/bd/0f/bd/1c/88/40/f0/DM00108282.pdf/files/DM00108282.pdf/jcr:content/translations/en.DM00108282.pdf).  Right now all peripherals are available even if your specific device doesn't have them.

almusil, 2020-04-23:
Hi,
AFAIK the pwm module is missing only PIN mapping for TIM3. The `TIM3` and `TIM2` are configured from the same registers that's why PAC [TIM3](https://docs.rs/stm32l0/0.10.0/stm32l0/stm32l0x1/struct.TIM3.html) points to the same register block as `TIM2`. 

IMO exposing the pins should be done under the  `io-STM32L071` feature. I can post a PR for that if you want. 

On the other support for the `TIM21` and `TIM22` is another story but still doable.

berkowski, 2020-04-23:
I ended up just using the PAC crate directly without `stm32l0xx-hal`.  Too many of my requirements were half- or unsupported that using the HAL as is just didn't make sense.