stm32-rs/stm32l0xx-hal #85: I2C 'Wait until byte is transferred' loop

ijager, 2020-03-18:
I was reading the `i2c` code since I want to implement it for another chip. Then I encountered the following loop in the `send_byte` function:

```
        // Wait until byte is transferred
        loop {
            let isr = self.i2c.isr.read();
            if isr.berr().bit_is_set() {
                self.i2c.icr.write(|w| w.berrcf().set_bit());
                return Err(Error::BusError);
            } else if isr.arlo().bit_is_set() {
                self.i2c.icr.write(|w| w.arlocf().set_bit());
                return Err(Error::ArbitrationLost);
            } else if isr.nackf().bit_is_set() {
                self.i2c.icr.write(|w| w.nackcf().set_bit());
                return Err(Error::Nack);
            }
            return Ok(());
        }
```
https://github.com/stm32-rs/stm32l0xx-hal/blob/f3b3e824252abfe6e5a592f3ce850d47d56cd112/src/i2c.rs#L188

This loop seems wrong as it only ever runs once. Guess it might still work in practice most of the time. I guess the idea was to wait for `txe` bit to clear like in two lines above? https://github.com/stm32-rs/stm32l0xx-hal/blob/f3b3e824252abfe6e5a592f3ce850d47d56cd112/src/i2c.rs#L183

dbrgn, 2020-03-18:
Most probably a duplicate of #77 and #76.

What version do you use? If master, does the commit before #76 was merged work? If you're using the released version, can you try master?

dbrgn, 2020-03-28:
@ijager can you test whether it works with the changes in #87 (https://github.com/gfroerli/stm32l0xx-hal/tree/i2c-correct-mappings)?

    # Cargo.toml
    stm32l0xx-hal = { git = "https://github.com/gfroerli/stm32l0xx-hal/", branch = "i2c-correct-mappings", features = ["rt", "mcu-YOURMCU"] }

closed 2020-04-05 by arkorobotics