stm32-rs/stm32l0xx-hal #92: Typo in serial.rs

tytouf, 2020-04-06:
I encountered what seems to be a typo in serial.rs when building with feature "io-STM32L021"

index 190c40d..7b46ea9 100755
--- a/src/serial.rs
+++ b/src/serial.rs
@@ -173,7 +173,7 @@ impl_pins!(
     PA0, AF6, LPUART1, RxPin;
     PA1, AF6, LPUART1, TxPin;
     PA2, AF4, USART2, TxPin;
-    PA2, AF6, LPUART1, TxPin,
+    PA2, AF6, LPUART1, TxPin;
     PA3, AF4, USART2, RxPin;
     PA3, AF6, LPUART1, RxPin;
     PA4, AF6, LPUART1, TxPin;

dbrgn, 2020-04-06:
That does indeed look like a typo! We should make sure that all 4 IO features are tested in CI :confused:

dbrgn, 2020-04-06:
I'll provide a fix for this case and the CI.

tytouf, 2020-04-06:
On a separate note I was able to toggle a pin and output on the uart very quickly: good job and thank you for the work on this crate!

dbrgn, 2020-04-07:
This should now be fixed on master, so the issue can be closed.

@arkorobotics maybe there should be a patch release on crates.io since 0.6.0 breaks compilation for some targets?

arkorobotics, 2020-04-09:
Done! Released v0.6.1.

closed 2020-04-09 by arkorobotics