stm32-rs/stm32l0xx-hal #103: serial doesn't seem to work at all

nebelgrau77, 2020-05-13:
I'm trying to use the serial example on STM32L031K6 Nucleo board. I'm getting the following compilation errors: 

```cargo
error[E0432]: unresolved import `stm32l0xx_hal::serial`: no `serial` in the root

error[E0599]: no method named `usart` found for struct `stm32l0::stm32l0x1::USART2` in the current scope

.usart(tx_pin, rx_pin, serial::Config::default(), &mut rcc)
          ^^^^^ method not found in `stm32l0::stm32l0x1::USART2`
```

in fact I can't find these in the documentation, but they clearly do exist in the actual src code in the repo. How can I get it to work?

dbrgn, 2020-05-13:
You probably didn't specify your MCU (see https://github.com/stm32-rs/stm32l0xx-hal/blob/b8147f13c455942b4374c353b33b486e23b30172/Cargo.toml#L97-L258).

For the STM32L031K6 you need `features = ["mcu-STM32L031K6Tx"]` or `features = ["mcu-STM32L031K6Tx"]`, depending on which MCU you have. (If you aren't sure which exact MCU you have, you can also use `features = ["io-STM32L031", "stm32l0x1"]`.)

dbrgn, 2020-05-13:
We should probably add a note about that in the README...

nebelgrau77, 2020-05-14:
Thanks! I was just missing the "io-..." part in the features block. Now it's working as expected :)

closed 2020-05-14 by nebelgrau77