stm32-rs/stm32l0xx-hal #106: Need more writers

arkorobotics, 2020-05-27:
@jonas-schievink, @lthiery, and @dbrgn:

For redundancy, I have added @hannobraun as a maintainer to this repository. I figured it's a good idea to have a backup person just in case. This action was purely made to give someone other than myself (who's also active in the embedded Rust community) access to the repository.

That said, your contributions to this repository have been critical to its growth and I would like to ask if any of you are interested in having Writer access. If you feel you'd make for a good maintainer as well, I'm open to that too. 

My goal here is to avoid getting in the way of progress because I'm too busy to review PRs.
 
I trust your judgement and believe the peer-review method we've been exercising thus far (i.e. - asking one another to sanity check code) is good enough for now. If things gets out of hand, we can always strategize a new plan. 

Please let me know your thoughts.

Regardless of your answer, I'm grateful for your contributions! Thanks!

dbrgn, 2020-05-27:
Thanks @arkorobotics for the offer :slightly_smiling_face: I'm happy to help with reviews and triaging of issues, however I still have to learn a lot about embedded software development (I'm a high-level software engineer by trade) and am not yet familiar with all the peripherals that ARM microcontrollers have. So feel free to add me to the project if you trust me, but I'd probably refrain from merging PRs myself :slightly_smiling_face:

arkorobotics, 2020-05-27:
Done! Welcome aboard! 

Funny enough, I made a similar point Hanno that I'm just an electrical engineer with a background in circuit design and robotics, and that this HAL is just a hobby to me. I guess we're both flying by the seat of our pants here!

hannobraun, 2020-05-28:
And I'm just some undifferentiated software developer who one day decided to specialize on embedded Rust, so... :-)

closed 2020-10-11 by arkorobotics