stm32-rs/stm32l0xx-hal #113: LPUART1 won't work with LSE as clock source

nebelgrau77, 2020-06-19:
Bottomline: baudrate not set correctly.

I've been trying to print a simple "Hello world" with serial, using LPUART1. I wanted to make sure that LSE was being used as the clock source, to make it truly low power, and I ran into a problem. I used the serial.rs example from the repo, changing the USART to LPUART, and before the clock setup I added the following code: 

```
dp.RCC.csr.modify(|_,w| w.lseon().on()); //enable LSE
    while dp.RCC.csr.read().lserdy().is_not_ready() {    
        // wait until LSE is ready
    }

dp.RCC.ccipr.modify(|_,w| w.lpuart1sel().lse()); //set LSE as LPUART1 clock source
```
If I do the exact same thing, but instead of .lse() use .system() or .ahb(), it works fine. With .lse() it just won't print anything. So I started checking registers in a GDB session and found out the following:
LSEON gets set correctly, and LSERDY is set as expected. Then LPUART1SEL is correctly set to 11: LSE selected as clock.

What doesn't look right is the value of LPUART_BBR, and I think I understand, why: the function that calculates the LPUARTDIV takes fck from the system clock, and that's why it all works if .system() is used. In my code the system clock is MSI running at 1.048 MHz. But to correctly set the baudrate I need the actual LPUART clock, so in this case 32kHz. 

I'm just pointing this out, I wouldn't really know how to fix it: I guess some match or if/else block based on the LSE/other choice. This is how CubeMX does it:

```

      case UART_CLOCKSOURCE_SYSCLK:
        lpuart_ker_ck_pres = HAL_RCC_GetSysClockFreq();
        break;
      case UART_CLOCKSOURCE_LSE:
        lpuart_ker_ck_pres = (uint32_t)LSE_VALUE;
        break;
```

I hope this will help someone who knows what their doing :) to improve this, meanwhile I'm trying to figure out how to go around it.

almusil, 2020-06-22:
The code in fact does not consider LSE as source for LPUART. This is why is it happening. There might be a couple of solutions. 

1) Having a trait implemented for LPUART only that will switch the source to LSE and reconfigures the baud rate correctly. 

2) Check which clock source we have and then use appropriate clock. That unfortunately does not fit nicely with current macro that generates the code. 

@hannobraun WDYT? The 2 seems too hacky IMO.

nebelgrau77, 2020-06-22:
Just to figure out how this was done for other chips I checked the HAL crates for F0xx, F1xx, F4xx and L4xx. Unless I missed something, none of them considers LSE frequency for baudrate divisor calculation, usually getting the fck from PCLK.

hannobraun, 2020-06-22:
I haven't worked with the LPUART, and I don't think I've made many changes to the clock code either (at least I don't currently remember many details about it). Since I also don't have the time to dig into it right now, any guidance I can give here is somewhat limited.

Generally speaking, I prefer trait-based approaches, as those can usually be implemented cleanly and allow the compiler to optimize out anything that's not needed for a given configuration. If anyone wants to tackle this, but can only come up with a more hacky solution, I'd merge that too. Better solve the problem and leave improvements for later than have broken code.

PR #131 (2021-03-02): Adds support of LSE clocking to LPUART
Fixes #113 

Based on PR #130
closed 2021-03-02 by hannobraun