stm32-rs/stm32l0xx-hal #117: Switching LSE on doesn't seem to work

nebelgrau77, 2020-07-24:
I'm trying to use LPUART with Low Speed External oscillator (LSE). According to the STM32L0x1 reference manual, you need to set bit 8 of the control/status register (RCC_CSR LSEON), and then check if bit 9 of the same register (LSERDY) is set. In my code I do this:
dp.RCC.csr.modify(|_,w| w.lseon().on());
but the bit is not being set, checked by reading the register in a GDB session. The same does work perfectly when a basic code is set up with STM32CubeMX/CubeIDE. I looked at the C code generated by CubeIDE, but there's nothing more than setting this bit, no waiting time or anything. Has anyone else experienced this problem? What am I doing wrong here?

akashihi, 2020-10-11:
The reference manual says that 'The LSEON, LSEBYP, RTCSEL,LSEDRV and RTCEN bits in the RCC control and status register (RCC_CSR) are in the RTC domain. ' Thus you need to disable write protection of RTC domain configuration.  This code works perfectly:

```
    p.RCC.apb1enr.modify(|_,w| w.pwren().enabled());
    p.PWR.cr.modify(|_,w| w.dbp().set_bit());
    p.RCC.csr.modify(|_,w| w.lseon().on());
```

nebelgrau77, 2020-10-11:
Great, thanks! Need to try it!

closed 2020-10-11 by nebelgrau77