stm32-rs/stm32l0xx-hal #118: .i2c method not found in `stm32l0xx_hal::pac::I2C1`

nebelgrau77, 2020-08-01:
I must be missing something, so please help: I'm trying to convert a working code, using SSD1306 display, from another STM32 board. I got stuck at setting up the I2C. It just won't compile, with the error as in the title. I tried the example from the crate, same problem. And when looking in the docs, I can't, in fact, find the .i2c method. After dp.I2C1. rust-analyzer shows only registers' names as possible methods. Am I missing something in the Cargo.toml that has to be set up?

dbrgn, 2020-08-01:
You probably need to pick the appropriate `io-*` Cargo feature, so that the HAL knows the proper function mapping of the I/O pins. The easiest way to do this is to use the `mcu-*` feature matching your MCU:

https://github.com/stm32-rs/stm32l0xx-hal/blob/d7f0e504bdb42af588cd1456c8e3b901766a0d45/Cargo.toml#L97

For example:

```toml
# Cargo.toml
stm32l0xx_hal = { version = "*", features = ["mcu-STM32L071KBTx"] }
```

(We really need to document this properly. I'll create a PR soon-ish.)

dbrgn, 2020-08-01:
@nebelgrau77 can you check out the updated README in https://github.com/stm32-rs/stm32l0xx-hal/tree/improve-docs and let us know whether it clears things up?

nebelgrau77, 2020-08-01:
OK, so that's what I thought, but I can't really get it to work. I did add the io-... to the features, but to no avail. Do I need to copy the whole set from the Cargo.toml in the examples, or is it enough to do this: 
```rust
stm32l0xx-hal = {version = "0.6.2", features = ["io-STM32L031", "stm32l0x1", "lqfp32","rt"]}
```
? 
Unfortunately it didn't work, just tried it :/

dbrgn, 2020-08-02:
What MCU are you using? The only two features you need are `rt` and `mcu-*` (the latter of course with your concrete MCU).

nebelgrau77, 2020-08-02:
It's an L031K6T6, 32 pins. The way I have it in the Cargo.toml right now is:

```toml
[dependencies]

stm32l0xx-hal = {version = "*", features = ["rt", "io-STM32L031", "stm32l0x1", "lqfp32"]}
stm32l0 = "0.10.0"
as-slice = "0.1.0"
embedded-hal = { version = "0.2.3", features = ["unproven"] }
cortex-m = {version = "0.6.1", features = ["const-fn"] }
cortex-m-rt = "0.6.8"
cortex-m-semihosting = "0.3.2"
void = { version = "1.0.2", default-features = false }
cast = { version = "0.2.2", default-features = false }
nb = "0.1.2"
```

I replaced the mcu- alias with the actual contents for my chip. Still no success :/

dbrgn, 2020-08-02:
Did you import the prelude or alternatively the `I2cExt` trait?

nebelgrau77, 2020-08-02:
Yep!

```rust
use stm32l0xx_hal::{pac, prelude::*, rcc::Config};
```

dbrgn, 2020-08-02:
Ok, then I'm out of guesses... Is this code public? If not, can you create an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

nebelgrau77, 2020-08-02:
yeah, I just put it here: https://github.com/nebelgrau77/stm32l0_i2c-problem/blob/main/src/main.rs - it's not my code, it's just the example code, with the references to the actual I2C device commented out, just to see if it would even compile. I'm sure I'm missing something in Cargo.toml, either too much or not enough :)

dbrgn, 2020-08-17:
I think you just made a typo that didn't exist yet in the original question :slightly_smiling_face: 

```diff
-    let mut i2c = dp.I2C1.I2c(sda, scl, 100.khz(), &mut rcc);
+    let mut i2c = dp.I2C1.i2c(sda, scl, 100.khz(), &mut rcc);
```

With this, and the following line in Cargo.toml:

```toml
stm32l0xx-hal = {version = "*", features = ["rt", "mcu-STM32L031K6Tx"]}
```

...the code compiles fine.

nebelgrau77, 2020-08-26:
Well that's really bizarre, but it did work this time, so that's great! Thanks!

closed 2020-08-26 by nebelgrau77