stm32-rs/stm32l0xx-hal #120: Register definitions for factory option bytes

rnestler, 2020-08-04:
I'm trying to use the internal ADC reference to monitor VDDA. To do that I need the `VREFINT_CAL` value from the factory options. The factory option location aren't available in the SVD file (probably because they aren't peripheral) so they are not accessible.

Would it make sense to add register like definitions to the HAL or the PAC for the factory option location?

For reference a section from the STM32L071xx datasheet:

[image]

and from the STM32L0x1 reference manual:

[image]

cc @dbrgn

dbrgn, 2020-08-05:
This field is part of the NVM bus:

[image]

I think it makes sense exposing factory option bytes and user option bytes through the HAL, but I'm not sure where the best place for that would be... Maybe an `nvm` module?

dbrgn, 2020-08-05:
@adamgreig @hannobraun are these fields exposed in one of the other HALs that you're aware of?

adamgreig, 2020-08-05:
F4 has it in `signature`: https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/signature.rs

dbrgn, 2020-08-05:
@adamgreig nice, thanks! That could probably be re-used for L0 as well.

Is the term "signature" commonly used for this type of information (calibration data)?

adamgreig, 2020-08-05:
I expect it came about because that module also covers reading the device unique ID (another random memory address you can't get from the SVD) and the device flash size, both of which are more 'signature'-esque. I haven't seen it for cal data but it makes sense alongside the other factory-written information I guess.

dbrgn, 2020-08-05:
Maybe the module could be called `nvm` or `factory` instead.

adamgreig, 2020-08-05:
Or `calibration` or `caldata`? But if you include the unique ID and flash size I think `signature` is OK, it still implies some sort of unique data about this particular chip.

That all said we also have https://github.com/stm32-rs/stm32-device-signature which provides access to unique ID and flash size (and is called `signature`) across many families including L0; perhaps the HAL could re-export that for unique ID and flash and then have a new module for cal data? I'm not sure about `nvm` since it's a bit generic and a `factory` module sounds like it should be a factory rather than contain factory data, though.

dbrgn, 2020-08-05:
Calibration data is provided in a new `calibration` module in #121.

Regarding the signature data, I'm not sure if it's worth it to pull in an external crate for [such a simple module](https://github.com/stm32-rs/stm32-device-signature/blob/master/src/lib.rs)... I'd tend towards copy-pasting the few lines into the HAL instead (with proper credits to @disasm of course). What do you think?

adamgreig, 2020-08-06:
No objections from me!

PR #121 (2020-10-16): Expose factory calibration data
This PR exposes the factory calibration data.

Verified to be working on an STM32L071KBTx. As a cross-check, I verified the memory addresses against the datasheet of the (randomly picked) STM32L053C6, they match.

[image]

Strongly inspired by the `signature` module in https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/signature.rs.

Fixes #120.

CC @rnestler
closed 2020-10-16 by hannobraun