stm32-rs/stm32l0xx-hal #128: serial module

pdgilbert, 2020-09-07:
Is there a reason that `serial` is not exported (pub ?) from `stm32l0xx_hal`?  When I build the `serial` example in the hal build directory it works fine, but if I run it in my own package I get
```
use stm32l0xx_hal::{pac, prelude::*, rcc::Config, serial};
                                               ^^^^^^ no `serial` in the root
```
Also, looking at  [the doc](https://docs.rs/stm32l0xx-hal/0.6.2/stm32l0xx_hal/) `serial` does not appear as a module. (Nor does `i2c` but I haven't tried that yet.)

pdgilbert, 2020-09-28:
I think there is something messed up in the feature / cfg setting of crate `stm32l0xx-hal`. In the crate build directory I can build example `serial` by specifying
```
cargo build --features="stm32l0x2 io-STM32L071"  --example serial
```
but not the seemingly more logical
```
 cargo build --features="stm32l0x1 io-STM32L071" --example serial
error: target `serial` in package `stm32l0xx-hal` requires the features: `stm32l0x2`, `io-STM32L071`
Consider enabling them by passing, e.g., `--features="stm32l0x2 io-STM32L071"`
```
Possibly I do not understand some naming convention.

However, even the combination that works in the crate's build directory does not work outside of the crate build directory. When cargo tries to resolve the crate version I get:
```
... depends on `stm32l0xx-hal`, with features: `io-stm32l021` but `stm32l0xx-hal` does not have these features.
```
In stm32l1xx/src/lib.rs I see:
```
#[cfg(not(any(feature = "stm32l0x1", feature = "stm32l0x2", feature = "stm32l0x3")))]
compile_error!(
    "This crate requires one of the following features enabled: stm32l0x1, stm32l0x2, stm32l0x3"
);
```
and also
```
#[cfg(any(
    feature = "io-STM32L021",
    feature = "io-STM32L031",
    feature = "io-STM32L051",
    feature = "io-STM32L071",
))]
pub mod serial;
```
If this really is all the way it is supposed to be? Could someone please share an example using the crate and `serial` outside the crate build directory? Also the necessary settings in Cargo.toml would be appreciated.

However, even if it works, I think the issue is that it is unnecessarily complicated. Can it not be made so that the crate can be used by specifying features indicating the HAL and the MCU, like  the other stm32*xx_hal crates do?

dbrgn, 2020-09-28:
> but not the seemingly more logical

Did you check out the updated README at https://github.com/stm32-rs/stm32l0xx-hal#supported-configurations? The best way is to use just the MCU feature for your MCU. Unfortunatel the "old" stm32l0x1/stm32l0x2/stm32l0x3 names don't cleanly map to the actual product categories.

> Can it not be made so that the crate can be used by specifying features indicating the HAL and the MCU, like the other stm32*xx_hal crates do?

Yeah, it should work by simply specifying your MCU (e.g. `features = ["mcu-STM32L071KBTx"]`). You can find an example project (which uses serial) here: https://github.com/gfroerli/firmware

Does that help?

pdgilbert, 2020-09-29:
Thank you, that helped a lot. I also see I was hopelessly optimistic in thinking there might be a simple naming convention. Obviously, I should have read the updated README, but I'm not sure I would have fully understood it without your project example. FWIW, I have several examples summarized at https://pdgilbert.github.io/eg_stm_hal/  and most of the `stm32l0xx-hal` versions now build.

closed 2020-09-29 by pdgilbert

dbrgn, 2020-09-29:
> Thank you, that helped a lot. I also see I was hopelessly optimistic in thinking there might be a simple naming convention.

So was I when I first started using this HAL :slightly_smiling_face: But - as always - it turns out reality is more complicated :slightly_frowning_face: