stm32-rs/stm32l0xx-hal #141: Support for the STM32L0x0 subfamily

nbraud, 2021-01-16:
Hi!

Support for the STM32L0x0 subfamily seems to be missing, according to both the README and `docs.rs`.
Presumably, this is because they are also missing from the Peripheral Access Crate, but what would be required (beyond some `svd2rust` calls) to make this happen?

I suspect @Jglauche and I would be willing to make it happen, if you can point us in the right direction.

nbraud, 2021-01-16:
Looks like the SVD descriptors are missing from the [files published by ST Micro]...  >_>'

[files published by ST Micro]: https://www.st.com/en/microcontrollers-microprocessors/stm32l0x0-value-line.html#cad-resources

jglauche, 2021-01-16:
I poked STM about it, will post an update if I hear from them.

jglauche, 2021-01-20:
They sent me
"STM32L0x0.svd Indeed, it seems the zip file is not updated on st.com, I will inform my colleague. For your information, you can also find this file at the following folder location: in STM32CubeProg tool (SVD folder) https://www.st.com/en/development-tools/stm32cubeprog.html or in STM32CubeIDE tool (\plugins\com.st.stm32cube.ide.mcu.productdb.debug_1.5.0.202011051456\resources\cmsis\STMicroelectronics_CMSIS_SVD)."

The file exists there, however it says it's under the "End User License Agreement for STMicroelectronics" license and I am unsure if this is any compatible to share so I asked them if this is okay to publish at this point

jglauche, 2021-02-03:
I received an updated today:
"Small update on my side.
We are taking this into consideration and we will try to deliver new SVD files with license indication for upcoming products.

Regarding you github question, you have to respect the following license:

Copyright (c) 2020 STMicroelectronics.

SPDX-License-Identifier: Apache-2.0

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License."

jglauche, 2021-02-03:
for easier access to the file until ST fixes their SVD download package, I created a repo with the file https://github.com/jglauche/STM32L0x0

snakehand, 2021-09-20:
There seems to be other differences as well that needs to be taken into account. I noticed that BP6, PB7 are reserved for USART2 in the data sheet for STM32L0101C6, but the current hal io implmentations place those pins on UART1 when building with the "closest" stm32l0x1 feature.

snakehand, 2021-09-28:
I tried making HAL support (specifically serial ports) for STM32L0x0 by hacking around in the stm32f0xx-hal crate. It seems a new io definition is needed as a number of peripherals are missing compared to the L0x1 devices. But still I am pretty stuck, as I can't get simple GPIO / UART code working. I can upload a fork of the repo with my hacks, but I doubt it has much value.

jglauche, 2021-09-29:
> I tried making HAL support (specifically serial ports) for STM32L0x0 by hacking around in the stm32f0xx-hal crate. It seems a new io definition is needed as a number of peripherals are missing [...]

Have you tried my fork at https://github.com/jglauche/stm32l0xx-hal ?

snakehand, 2021-09-30:
When building for the chip on my dev board, using this feature set : stm32l0xx-hal = { path="../test/stm32l0xx-hal", features = ["mcu-STM32L010RBTx", "stm32l0x0", "rt"] }  - I get a ton of compilation errors, which I think are triggered by the wrong io feature. "io-STM32L071" pulls in a bunch of perpherals that don't exist in the L010 HAL.

nikgul, 2022-09-17:
Is there any chance of this getting included ?
I'm new to rust, working through "The book" atm. so I don't know how much help I would be able to provide.