stm32-rs/stm32l0xx-hal #161: adc to internal temperature

pdgilbert, 2021-04-04:
Is it possible to read the internal temperature with adc. I think it is connected to internal channel ADC_IN18 but I don't see how to specify that. (Newbie, so I may just be doing something stupid.)

hannobraun, 2021-04-06:
The internal temperature sensor is represented by `adc::VTemp`. I'm not sure how to use it, but it has an `enable` method, and presumably can be treated like any other channel after that.

It would be great if we had an example showing this in the repository.

pdgilbert, 2021-04-06:
Ok, I got it to build. I just used VTemp as the channel and didn't need to do any `enable` for it to build. I don't have hardware to test that this actually works. I can do a PR with an example if you like,  but not testing on hardware is a caveat.

hannobraun, 2021-04-07:
Well, I guess having an example that points users in the right direction would be valuable, even if it's not tested. I'd merge your PR, as long as the example has prominent comments that clearly state it needs testing and might possibly not work.

pdgilbert, 2021-04-07:
I will do a PR shortly. In addition to saying it has not been tested I will suggest if anyone does test it then they could report results here (issue #161). I'll also add a comment about my site https://github.com/pdgilbert/eg_stm_hal/ which has `setup()` functions for other devices, and several other examples too. Let me know if you would prefer that I omit that.
Thanks for your help.

closed 2021-04-07 by pdgilbert