stm32-rs/stm32l0xx-hal #163: Clippy CI build fails

hannobraun, 2021-04-08:
The Clippy CI build currently fails, presumably because a new Clippy version has been pulled in. It would be nice to get this fixed (either by following what Clippy says or by silencing those warnings; haven't looked into it, so don't know what would be appropriate). Any pull request doing so would be very welcome!

hannobraun, 2021-04-08:
Here's a failing build: https://github.com/stm32-rs/stm32l0xx-hal/runs/2296253088?check_suite_focus=true

dbrgn, 2021-04-08:
Regarding the non-uppercase identifiers: I'd suggest silencing these warnings since acronyms are used a lot. Simply applying the fixes might break third party code without any gain.

I'll give it a try.

PR #164 (2021-04-09): Fix clippy warnings
This should hopefully fix all issues reported by Clippy.

Resolves #163.
closed 2021-04-09 by hannobraun