stm32-rs/stm32l0xx-hal #166: update PAC from 0.10.0 to 0.13.0

allexoll, 2021-04-30:
there has been updates to the pac. 

some fields were missing (namely in the CRC peripheral, and i'm porting the F4 CRC driver to this family, but some fields aren't present in the 0.10.0 pac. 

trying to change the pac version only seems to break 2 calls to unsafe functions in the DMA driver, so that seems fixable.

What is your opinion on this?

```
error[E0133]: call to unsafe function is unsafe and requires unsafe function or block
   --> /Users/allexoll/Desktop/stm32l0xx-hal/src/dma.rs:344:55
    |
344 |                       handle.dma.$chfield.par.write(|w| w.pa().bits(address));
    |                                                         ^^^^^^^^^^^^^^^^^^^^ call to unsafe function

error[E0133]: call to unsafe function is unsafe and requires unsafe function or block
   --> /Users/allexoll/Desktop/stm32l0xx-hal/src/dma.rs:351:55
    |
351 |                       handle.dma.$chfield.mar.write(|w| w.ma().bits(address));
    |                                                         ^^^^^^^^^^^^^^^^^^^^ call to unsafe function
```

hannobraun, 2021-05-01:
Replied in #167.

allexoll, 2021-05-04:
has been adressed in #168

closed 2021-05-04 by allexoll