stm32-rs/stm32l0xx-hal #198: cortex-m-rt v0.7.0?

pdgilbert, 2021-12-05:
Any chance of updating to use `cortex-m-rt v0.7.0`? That may also need updating to `stm32l0 v0.14.0` and other changes.

If anyone can point me to a branch that has this update then I can reactivate testing my examples (https://github.com/pdgilbert/rust-integration-testing/actions) with `stm32l0xx-hal`. I have had to disable it because of the conflict with other crates now using the newer `cortex-m-rt`.

jamwaffles, 2021-12-06:
I did the upgrade in a fork of this repo a while back, but found a regression with the timers. I opened https://github.com/stm32-rs/stm32-rs/pull/659 to fix, but there's been no movement on that so far.

kuon, 2022-06-16:
Shouldn't it be OK to upgrade now as the mentioned issue seems to be fixed?

jamwaffles, 2022-06-16:
That PR was merged some time ago, but there hasn't been an stm32-rs release since, so we need to wait for that.

PR #215 (2022-08-15): Upgrade `stm32l0` to v0.15.1
Mostly mechanical changes resulting from the `stm32l0` crate upgrade.

I also:

1. Removed `cross` from CI as it was breaking things and I don't see the value in using it
2. Upgraded to a newer RTIC for the examples that use it

Closes #198.
closed 2022-08-15 by jamwaffles