stm32-rs/stm32l0xx-hal #204: docs.rs failed to build the docs for v0.8.0

azerupi, 2021-12-09:
Docs.rs failed to build the documentation for v0.8.0: https://docs.rs/crate/stm32l0xx-hal/0.8.0/builds

From the logs:
```
[INFO] [stderr]    Compiling stm32l0xx-hal v0.8.0 (/opt/rustwide/workdir)
[INFO] [stderr]    Compiling quote v1.0.10
[INFO] [stderr]     Checking num-integer v0.1.44
[INFO] [stderr] error: failed to run custom build command for `stm32l0xx-hal v0.8.0 (/opt/rustwide/workdir)`
[INFO] [stderr] 
[INFO] [stderr] Caused by:
[INFO] [stderr]   process didn't exit successfully: `/opt/rustwide/target/debug/build/stm32l0xx-hal-5ede552dba319221/build-script-build` (exit status: 101)
[INFO] [stderr]   --- stderr
[INFO] [stderr]   thread 'main' panicked at '
[INFO] [stderr] 
[INFO] [stderr]   Must select exactly one flash size for linker script generation!
[INFO] [stderr]   Choices: 'flash-8', 'flash-16', 'flash-32', 'flash-64', 'flash-128' or 'flash-192'
[INFO] [stderr]    Alternatively, pick the mcu-feature that matches your MCU, for example 'mcu-STM32L071KBTx'
[INFO] [stderr] 
[INFO] [stderr]   ', build.rs:43:13
[INFO] [stderr]   note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[INFO] [stderr] warning: build failed, waiting for other jobs to finish...
[INFO] [stderr] error: build failed
```

This is caused by the exclusive MCU selection. I'm not sure if there is a great way to handle mutually exclusive features on docs.rs?

I think the best way would be to add `#[cfg(any(feature="xxx", doc))]` to enable everything in the doc generation. (https://doc.rust-lang.org/rustdoc/advanced-features.html#cfgdoc-documenting-platform-specific-or-feature-specific-information).

Then in the future when it is stabilized we should add `#[doc(cfg(feature="xxx"))]` to mark some items as only available using certain features. https://doc.rust-lang.org/unstable-book/language-features/doc-cfg.html

dbrgn, 2021-12-09:
Already fixed in #203. I'll create a new patch release.

I'd wait with additional config flags until the doc-cfg syntax is stabilized.

closed 2021-12-09 by dbrgn

dbrgn, 2021-12-09:
I'll push out the new release as soon as #205 is merged.

azerupi, 2021-12-09:
Sorry I missed that, thanks for the quick reaction :+1:

dbrgn, 2021-12-13:
Release 0.9 is out: https://docs.rs/stm32l0xx-hal/0.9.0/stm32l0xx_hal/