stm32-rs/stm32l0xx-hal #211: Stepping down as a maintainer

hannobraun, 2022-01-12:
Hey folks!

I'd like to announce that I'm stepping down as a maintainer of this crate. [I'm off to new adventures](https://github.com/sponsors/hannobraun) and would like to be able to focus on this new direction going forward.

Since that means that @dbrgn is the only remaining active maintainer, this might be an opportunity to look for new maintainers. But that is not my choice to make. I'm will be removing myself from @stm32-rs/stm32l0 shortly after posting this.

cc @arko, @dbrgn (the other members of @stm32-rs/stm32l0)
cc @adamgreig (chief of stm32-rs)

jamwaffles, 2022-01-12:
I'd be happy to take a more passive maintainership role for this crate. I'm not skilled enough to make architectural decisions but I can help out on the easier feature/bugfix PRs.

hannobraun, 2022-01-12:
> I'd be happy to take a more passive maintainership role for this crate. I'm not skilled enough to make architectural decisions but I can help out on the easier feature/bugfix PRs.

Honestly, I think that's all that's required to keep things going. I myself have been leaning heavily on contributors for knowledge and testing for a while now. Providing a second pair of eyes and pressing "Merge" in a timely manner is already a very valuable role.

jamwaffles, 2022-01-12:
In that case, count me in!

[image]

dbrgn, 2022-01-12:
Oh, that's sad to hear, @hannobraun. Thanks for all the maintenance work you've done here! However, your CAD project sounds very interesting, good luck 🙂 I'll definitely check it out.

> I'd be happy to take a more passive maintainership role for this crate. I'm not skilled enough to make architectural decisions but I can help out on the easier feature/bugfix PRs.

Same thing for me 😄 As noted in #106, I'm a high-level software engineer and many things in the embedded world are still black boxes to me (for example, I've never used DMA so far and some aspects of timer configuration are still mysterious to me).

I think in the long term it would be good to have someone with more embedded experience that would be able to properly review pull requests. Until then, we could adopt a more passive maintenance model, where we generally accept all pull requests that are well-documented and don't look totally wrong, making the submitters responsible for their code. This could be documented in a more explicit way (e.g. with a note in the README), also with a call for more maintainers.

> Since that means that @dbrgn is the only remaining active maintainer, this might be an opportunity to look for new maintainers. But that is not my choice to make. I'm will be removing myself from @stm32-rs/stm32l0 shortly after posting this.

I'd add @jamwaffles as co-maintainer, but I don't actually have the permission to edit members of that group (nor do I have maintainer-permissions for this repository, for that matter). @hannobraun do you know whom to ping?

hannobraun, 2022-01-12:
Thanks for the kind words, @dbrgn!

> I'd add @jamwaffles as co-maintainer, but I don't actually have the permission to edit members of that group (nor do I have maintainer-permissions for this repository, for that matter). @hannobraun do you know whom to ping?

Are you sure? When I was still in the team, it looked like I could add members on [the team members page](https://github.com/orgs/stm32-rs/teams/stm32l0/members), but since I didn't actually do it, I either might have been mistaken, or maybe I was a maintainer and you're a normal member? I'm not sure; not an expert on GitHub org stuff.

If you really can't do anything on that page, @adamgreig should be able to help. Besides adding @jamwaffles to the team, it might be a good idea to make one or both of you maintainers in that team, while @arkorobotics is AWOL.

> (nor do I have maintainer-permissions for this repository, for that matter)

I thought any rights we have here were conferred to us through our team memberships, but it seems I was wrong. At the very least, I can still open the repository settings (although I can't seem to do a whole lot there). And I still can close random issues.

Looks like I didn't actually step down. I just removed the privileges that would have been useful to add @jamwaffles, but kept all the others :laughing:

adamgreig, 2022-01-12:
I've made @dbrgn a maintainer of the stm32l0 team, so you should now be able to add new members. Anyone in the team has maintainer access to this repository, but it looks like you were also directly listed on the repository as a holdover from before... I've cleaned up the direct repository access so it should all be through the team now.

Thanks for all your hard work here too, @hannobraun!

hannobraun, 2022-01-13:
Thank you, Adam!

dbrgn, 2022-01-13:
Thanks @adamgreig!

I can now add people to the team, but only if they're already part of stm32-rs. I cannot add people to stm32-rs, so unfortunately I'm still unable to add @jamwaffles 🙂

adamgreig, 2022-01-13:
Oh, that's annoying! I don't think I can enable that, so I've invited @jamwaffles to the org myself. Please let me know whenever anyone else needs inviting (and welcome, @jamwaffles!)

jamwaffles, 2022-01-13:
I'm in! Thanks everyone, and thank you @hannobraun for your stewardship of this crate :)

dbrgn, 2022-01-13:
Perfect, thanks @adamgreig!

dbrgn, 2022-01-13:
Call for maintainers added in #212.

closed 2022-01-13 by dbrgn

arkorobotics, 2022-01-14:
@hannobraun I'm still alive. Looks like @adamgreig took care of things here (Hi Adam!). I thought I'd chime in to say how great it has been to watch this repo grow over the years. It wouldn't be where it is today without you! Best wishes on your new adventures with Fornjot!

hannobraun, 2022-01-14:
Thank you, @arkorobotics!

allexoll, 2022-01-23:
I'd be happy to help as well, I've been using l0xx as my go-to m0 lately, if help is still needed!

dbrgn, 2024-01-16:
A quick update from my side: I'm not actively using stm32l0 MCUs anymore, so I will only rarely be of help. Don't wait for me on any reviews 🙂 Only if there are smaller maintenance tasks or small PRs that don't require any testing with an actual MCU, I might be able to help out from time to time.

PR #212 (2022-01-13): README: Add call for maintainers
I guess that should do. What do you think, @jamwaffles?

Closes #211.